nathanRamaNoodles/Noodle-Synth #13: issue with init variable when calling 'newSong'

wEPac, 2019-04-28:
Hello, thanks a lot for this great library, I tried to reduce code size and I found some tiny bugs, in the function `newSong(const char *p)`, when looking for flat and sharp, the both arrays arent properly initialized. If the new song doesnt have global flat or sharp, the last setting will be held. The arrays should be initialized at the begining of the function. Sae about the name of the song. Something like this :

`MusicWithoutDelay& MusicWithoutDelay::newSong(const char *p)
{
  default_dur     = 1;  //eqivalent to a wholenote
  default_oct     = 4;
  bpm             = 100;
  wholenote       = (60 * 1000L / bpm) * 4;
  
  playSingleNote  = false;
  start           = true;
  mySong          = p;
  loc             = 0;
  memset(songName, 0, SONG_NAME_LENGTH);
  memset(autoFlat, 0, 5);
  memset(autoSharp, 0, 5);
  slurCount = 0;
  

  // get the name
  if (pgm_read_byte_near(mySong) != ':')
  {
    ...`

nathanRamaNoodles, 2019-04-28:
Thank you for joining my little hobby project. I thought it was a lost cause, but at least you're having fun. Some stuff I wanted to do was the ability to read midi files and play them; but I decided to just turn the Arduino into a synth engine(so I would send midi data to the rx and tx lines) 

Hmm. Interesting. I forgot about memsetting the flats and sharps.  However, I thought the name would be renamed properly.
Also, reversing is something I was not able to make efficient.  I got stuck a year ago because when you read integers backwards, building a number from chars is different than reading it forward.

wEPac, 2019-04-28:
Really your library is the best I found. I just found it 1 second before to write my own one, you saved a lot off precious time for me. The ability to read midi will probably tickle me one day, actually I need the RTTL for my clock project ;-) I made a fork where I did some modifications in the `MusicWithoutDelay& MusicWithoutDelay::update()` function. I made 2 new private functions to manage this part, and when reading in reverse, I just make the `loc` increment pointer to jump at the begining of the note, so we will read it easy. I maybe found a better way to read flat and sharp too, this is working fine and it allows a global sharp to be depreciated if we add a flat to the note. Actually I saved more than 2kB, we never have enough on our tiny MCU :-P I m writting a little sketch to show a RTTL song like a ppartition, this is helpful to find mistake when writing a RRTL string. I will publish it soon, feel free to add it in your library examples if you like it.

wEPac, 2019-04-28:
Ah LOL, I m too much tired, I didnt see you already looked at the fork, since you did, I came back to the v3.5 but this part about reversing is very close. Also, since yesterday I finished to rewrite the reverse part. I also made some change for the wave table ;)

nathanRamaNoodles, 2019-04-28:
Wow thank you for your contributions.  I can't wait to see my little baby running on lite memory.

The only difference in v4.0.0 is the global synthEngine and the mixture of midi instruments with MusicWithoutDelay objects.
Glad, your enjoying your journey in c++ and memory management.

Also, get a teensy 3. You will be amazed at the number of midi instruments this baby can run at.

wEPac, 2019-04-28:
hehe yes, teensy 3 is making me drooling, especially for the FFT spectrum analyser I made, I could reach a sampling of 16kHz when Nano hardly reach 6kHz. But that s great fun to work with limited features, to push them at the limits, and nano is under 3€ :-D

About the change I made, 2 are a bit destructive:
 - the noisy wave table is reduced to 3/4 and repeated, but not in a linear form, so the effect is good.
 - the Envelopes are reduced by 4 times too, I didnt noticed real difference, but maybe it is better to keep them with full definition...

nathanRamaNoodles, 2019-04-28:
Some issues i got from your fork:

+ flats/sharps are wrong
   + I tried one of my examples: [Zelda_New_version](https://github.com/wEPac/Noodle-Synth/blob/master/examples/MusicWithoutDelay/songs/Zelda_New_version/Zelda_New_version.ino)
And I hear one of the flats/sharps are wrong.  Perhaps something is off?  
+ The other voices are not playing in sync
   + I tried my [Spider_Dance_Undertale program](https://github.com/wEPac/Noodle-Synth/blob/master/examples/MusicWithoutDelay/songs/Spider_Dance_Undertale_new/Spider_Dance_Undertale_new.ino) and the Arduino only plays the melody while the other instruments are only played at the beginning and then drop out later on.  Strange...

I see you changed a lot of files including `_skipSolver()` so it may take a while for me to crack this bug; `skipSolver()` was the hardest feature for me to figure out since Arduino runs in a single thread without delay.  
Anyway, if you need help, I'll be here.  And if you have any questions on how the `SynthEngine` works, I'll be more than happy to explain it. (TBH, it took me around 4 months to build the engine and add support for multiple boards)

Have fun  :)

nathanRamaNoodles, 2019-04-28:
Also, you probably know this, but I'll say it here anyways for others:
Replace the examples' dependencies from 
```
#include <NoodleSynth.h>
synthEngine mixer(20E3);
```
to
```
#include <MusicWithoutDelay.h>
```
And in Arduino's list of libraries, delete my NoodleSynth so that it won't conflict with your fork.

wEPac, 2019-04-29:
Hello hello ;)
I changed the name only for `skipSolver()` to `_skipSolver()` just to be neat, like this is a private function, but nothing (yet?) in it ;)
I will check further for the flat/sharp, I have to give a try with the Zelda song. The tool to print the partition is almost done, this will help to debug.
I've to check this issue about synch, but right now I am ppuzzled. I only rewrote how to read note. However I got something strange sometimes, when burning the sketch into the MCU, I sometimes had this issue which has disappeared when I burn it again. Something like the IDE had an issue and I had to restart IDE to have cleen index? I will check further ;)
Glad to know about to replace the dependencies, maybe a setting with a `#define` could be useful to select? Anyway, soon I fixed these issues, I will look further for your Engine, no doubt :-)

wEPac, 2019-04-29:
Kinda fun, about Zelda and the #, like you defined the sketch we can only have 5 global sharp notes, and so, the e# never exist, neither the b#. The abusive notation of a b# to play a c is the issue. In your global settings for Zelda, you have f=aeb..., We could consider to play a b# like a c, but in this case, we must change `autoSharp[5];` into `autoSharp[7];` and I guess to modify the code, it never played a b# ;-)

However, like this notation is abusive, I think the code shouldnt be changed. If we have to check for each mistake or abusive notation, the code size will increase for nothing.

wEPac, 2019-04-29:
I gave a try to Spider Dance, it looks to play fine with the current modifications.

However I m still working on the code, I rewrote `_skipSolver();` and `skipTo();`, I am checking further for pointer issues, but I m going to publish the actual code if you want to check it ;)

nathanRamaNoodles, 2019-04-29:
Yeah, I'm interested. I'll post a audio clip later today about the spider dance. Mine sounds different than your fork.  It's a small difference as soon as `skipto(0)` is called.  I'll check your `skipto()` function.

nathanRamaNoodles, 2019-04-29:
Also, in music I recall an `a#` is the same as `b_`.  This happens sometimes in a music sheet when the author wants to pull another note that doesn't follow the typical major or minor scale.

So forcing a `a#` in a scale with only a `a_` is user friendly to the programmer and the musician.  At least, that's what I think.

wEPac, 2019-04-29:
Zelda looks to work fine as well, checking further I added the abusive notation about b# and c_ to work.

wEPac, 2019-04-29:
still a minor issue with `skipTo`, looks like the pointer is going back into the header, I m working on

nathanRamaNoodles, 2019-04-29:
Take your time, we only got our whole lives :) .  Also, are you in college by any chance?  I'm only 20 right now.  Are you interested in Electrical engineering, computer engineering, or computer science. Basically, what do you want to do in life? 😅

I don't know where I wanna go.  I mean I've done some software stuff, but I haven't made any money with it.

P. S. 
Making money with Arduino is a waste of time and near impossible.

wEPac, 2019-04-29:
corrected a silly bug in `skipTo` where I used the pointer `loc` when `skipCount` must be used

wEPac, 2019-04-29:
I m not hurry but passionate, programing is a very exciting mind game :-P. I like also good to electric stuff, than computer, or to split apart my motorbike motor on the kitchen desk (GF doesnt LOL) Upon that I m an old 'monkey' and I m looking for fun and pleasant people, far away any kind of work (I feel work like slavery). Right, making money is not easy, especially with every kind of thing that can be copied. The reason why, I sommetimes do things under GNU license, and sometimes it will stay commercial. What I noted, many people will consider your work, depending of the price you will sell it. Ya! that s so stupid but right, you can sell a piece of shit if the price is 1000$. A marvel you would sell for cheap, nobody will want it. 

A good way to earn some money is to make scripts in some games like Second Life, cause they are protected and good scripters are rare there ;-)

wEPac, 2019-04-29:
take a look into my fork, at 'example/tools', this is not finished but already working, download the both files...

nathanRamaNoodles, 2019-04-30:
Sorry it took long to respond, I was working on Home Automation at my church.  I couldn't wait to try out your code.
```
What I noted, many people will consider your work
```
I'll make sure to add your name in the contributions at the top of the ReadMe file. 😄 
```
you can sell a piece of shit if the price is 1000$. A marvel you would sell for cheap, nobody will want it
```
Finally, someone I can relate to.  Most Arduino programmers are noobs who don't know how to program.  But once I saw your code, I realized I can relate to you.  I love to program as a passion; like this project.  But to get a job is to show off our open-source projects.  It's a tough world out there, so thank you for posting this :)

```
take a look into my fork, at 'example/tools', this is not finished but already working, download the both files...
```
Oh my.  I tried your example and it sounds amazing and so musical ❤️ .  This is better than I could even write.  Your coding skills are very professional and readable.  I especially like the first song: `M Jarre - Lara's theme`.  I'll definitely try porting this over to mine.  
I also can't believe you made an RTTL to music sheet converter, otherwise known as a "partition".  😃 
That must have taken a while.  👍 

However.............I did notice a few things off:

1. The  [spider Dance's](https://github.com/wEPac/Noodle-Synth/blob/master/examples/MusicWithoutDelay/songs/Spider_Dance_Undertale_new/Spider_Dance_Undertale_new.ino) first instrument drops out.  Actually, I am only able to hear two voices.  So is the library stuck somewhere after two MusicWithoutDelay objects?
2. Playing the music backward is off by a few milliseconds.  But to be honest, you can get rid of that function.  Playing music in reverse isn't really popular anyway.  This can save us a lot of code too.

Other than that, I'd say your fork is perfect. Good job 👍

wEPac, 2019-04-30:
Welcome back, I just woke up here :D, thank you for these nice comments, I m glad you appreciated, this is one of the meaning too, when sharing code, to know this can be useful and people will get fun with it ;)

Your library is great, see how it sounds just coding some traditional music sheet! At one moment I was wondering to use a MP3 player for my clock but finaly, I will use RTTL, pure coding project is funiest hehe.

I am still cleaning the code, trying to remove some unused part or declaration... Reverse is interesting and it will work, be sure, I am stubborn hehe, and the last issues now are just tiny bugs about pointers.

I am still wondering why, I didnt get the reason, during the `update()`, before and after reading the note letter, you have `if (!skip)...` and then to read for duration and twice time for a dot?

When the basic will work I will look further for issues like with Spider dance, and I will have a question for you, is there a possibility to make a function `autoPlay()`, a possibility for the timer interupt to call an update when the note is played and done?

nathanRamaNoodles, 2019-04-30:
```
Reverse is interesting and it will work, be sure, I am stubborn hehe
```
Oh, that gives me future headaches 😄 
Also, there is a really cool effect to make a reverse sound effect.  Call `instrument.overrideSustain(true)`  and then `instrument.setSustain(REV_SUSTAIN)`  ;)

```
I am still wondering why, I didnt get the reason, during the update(), before and after reading the note letter, you have if (!skip)... and then to read for duration and twice time for a dot?
```
I must have been very deep in my code.  I was in God mode 😄 .  I kinda forgot, but I remember it helped prevent overlapping of notes.  For instance, 
```
instrument 1:          a                    b                    c
instrument 2:          a          b                              c
skipTo(5) :                              *
```
If we skip to time location "5 seconds", the `if(!skip)` will purposely skip notes to make both instruments start with the note closest from the left of the time location.  So, at time "5", the Arduino will set instrument 1 to note "a" and will set instrument 2 to note b.  This is extremely complicated because we are dealing with a microcontroller who can't do processes in parallel.  So I basically...made is done in series 😓.  But it's fun.  TBH, I forgot exactly how `skipTo` works, but I remember parts of it.

```
is there a possibility to make a function autoPlay(), a possibility for the timer interupt to call an update when the note is played and done?
```
You can poll `instrument.isEnd()` in `loop()` and then if you want to play a song once or three times call, `instrument.play(1)` or `instrument.play(3)`  Interrupts aren't possible(I think), because timer 2 is used for sound, and timer 1 for manipulating that sound with wavetables.

```
This is also a good way to introduce to C++ and programation, so, we cannot complain about newbies hehe, we all have been newbie
```
Yeah, my first project was blinking an LED ;P.  My almost complex project was this library's synth engine and `skipTo()` function.  You probably know why 😄.  This project taught me about memory management and pointers...and mixing audio...and GitHub.

wEPac, 2019-04-30:
ah yes, I was just wondering how to synch 2 instruments after a `skipTo()`... now this have a meaning, actually it will start at a full note, I ve to preserve your idea.

Life would be boring if we couldnt learn or progress hehe.

about an `autoPlay` I mean without to call an `update` for sure. Interrupt are busy, like you said, but synth is using one to manage a note, maybe here, we could check when it could call a next update? This will be very powerful to play a music that way, I think this should be a very great add on at your library

nathanRamaNoodles, 2019-04-30:
> now this have a meaning, actually it will start at a full note

I believe `skipSolver()` figures out the amount of time left to play the "leftover" note. That part was hard too; too much math in my head. In my code, go to Zelda and go in the serial monitor.  Then try pausing, reversing/forwarding, and skipping with brute force; I guarantee you it will stay in sync 😄 
;)

> but synth is using one to manage a note, maybe here, we could check when it could call a next update?.......This will be very powerful to play a music that way, I think this should be a very great add on at your library

Hmm, I'll research this.  Seems interesting 👀

wEPac, 2019-04-30:
yes, that s it, `skipSolver()` will send the duration for a note pointed by `skipCount`

`skipTo()` will first count time to reach the nearest note for the time we want.
I guess, soon we found the note and set the pointer `loc` to the concerned note, we have to manage `millis()` to keep the synch and to play only a part of the note duration

wEPac, 2019-04-30:
hello, may you try this?....
I noticed sluring notes werent really sluring and most of times, we have tiny horrible 'clack' between notes with no sustain. I did a try to 'round' lightly the end of notes with no sustain. It s giving nice sluring and reducing 'clacks', beat is preserved but the whole feeling is smoother.

If you could try it and to tell me your opinion? Changes are into `synth.cpp`. I changed this in the `Volume envelope generator` part : 
```
        case NONE:
          value  = (value * -1) + volume[divider];
          value += (((unsigned char*) & (EPCW[divider] += EFTW[divider]))[1]);
          break;
```
into

```
        case NONE:
          // even monotone note must be lightly rounded (prevent 'clacks')
          if (((unsigned)value) > 24)   // sounds like best results from 16~32
          {
            value  = (value * -1) + volume[divider];
            value += (((unsigned char*) & (EPCW[divider] += EFTW[divider]))[1]);
          }
          break;
```

nathanRamaNoodles, 2019-05-01:
The Sustain NONE should have no rounding because its supposed to sound like a buzzer with no volume decreasing/increasing over time.  For slurring, you are right; I need to add a new `case 'SLUR':` that will handle slurring.   Actually, a year ago, I didn't know how to slur.  A slur is simply the volume is carried through multiple notes.  So for instance, 
```
a+b+c, d
```
the instrument will start loudly at `a` and then decrease volume throughout `b` and `c`.  And then at `d` is triggers at full volume.
Volume is also a tough subject in Audio ;)
So NONE should be the same as in the master branch, and I need to figure out SLUR next.

> It s going to work fine but I detected a bug when using pause. Sometimes a track will stop to play, I suspect an issue with a timer value, probably a > when a >= would match, I m looking for.

I'm not sure.  I never have issues with `pause()`.  Could you give an example sketch for testing it?

> Another question, Envelope2 is really making sound volume low, may you quickly explain to me how the envelope is working?

Because the attack of the note is halfway.  So the volume doesn't have enough time to reach there.  I personally like `ENEVELOPE0` and `ENEVELOPE1`.  Envelopes are what basically allow us to control volume over a time without using another Timer.  It's basically a hack I learned from [dzloine](https://github.com/dzlonline/the_synth)

I like these questions. It makes me think about my masterpiece a year ago:)

wEPac, 2019-05-01:
arggg, found a deadly bug with envelopes
```void synth::setEnvelope(uint8_t voice, uint8_t env)
{
  switch (env)
  {
    case 1:
      envs[voice] = (unsigned int)Env0;
      break;
    case 2:
      envs[voice] = (unsigned int)Env1;
      break;
    case 3:
      envs[voice] = (unsigned int)Env2;
      break;
    case 4:
      envs[voice] = (unsigned int)Env3;
      break;
    default:
      envs[voice] = (unsigned int)Env0;
      break;
  }
}```
case  will never happen, the code should be:
```void synth::setEnvelope(uint8_t voice, uint8_t env)
{
  switch (env)
  {
    case 0:
      envs[voice] = (unsigned int)Env0;
      break;
    case 1:
      envs[voice] = (unsigned int)Env1;
      break;
    case 2:
      envs[voice] = (unsigned int)Env2;
      break;
    case 3:
      envs[voice] = (unsigned int)Env3;
      break;
    default:
      envs[voice] = (unsigned int)Env0;
      break;
  }
}```

nathanRamaNoodles, 2019-05-01:
Lol, silly mistake. Thx

wEPac, 2019-05-01:
> The Sustain NONE should have no rounding because its supposed to sound like a buzzer with no volume decreasing/increasing over time. For slurring, you are right; I need to add a new case 'SLUR': [...]

mmm I think you are right, I will add the slurring I made, we can always modify it later, so `NONE` will not change. The slurring is only used by the update actually ;)

wEPac, 2019-05-01:
I just made an update (for the sketch to writ music sheet as well), it should be close now, I removed unused variable and also redefined many they were too big (many just needed 8 bit) also removed all float variable and constant.

I gave a try to envelopes, modifying just the first 1/4, so the volume is similar for all envelopes. I guess this could be a deal because it is balanced between envelope effect and what we can hear...

wEPac, 2019-05-02:
found an easy way to make `slur`, I changed this:
```
  int set = 0;
  if      (duration >= 1000)  set = 105;
  else if (duration >=  400)  set = 87;
  else if (duration >   300)  set = 82;
  else if (duration >   250)  set = 78;
  else if (duration >   200)  set = 73;
  else                        set = 55;
  edgar.setLength(myInstrument, set);

...

  if (!sustainControl)  // user did not define sustain, we are free to use it
  {
    if (pgm_read_byte_near(mySong + loc) == '+')  // a slur note, "+"
    {
      setSustain(NONE);
    }
    else              // a ',' or NOTHING for the last note of the song
    {
      setSustain(SUSTAIN);
    }
  }
```

into

```
  if      (duration >= 1000)  noteDur = 105;
  else if (duration >=  400)  noteDur = 87;
  else if (duration >=  300)  noteDur = 82;
  else if (duration >=  250)  noteDur = 78;
  else if (duration >=  200)  noteDur = 73;
  else                        noteDur = 55;

...

  if (!sustainControl)  // user did not define sustain, we are free to use it
  {
    if (pgm_read_byte_near(mySong + loc) == '+')  // a slur note, "+"
    {
      noteDur += 10;
      setSustain(NONE);
    }
    else              // a ',' or NOTHING for the last note of the song
    {
      setSustain(SUSTAIN);
    }
  }
  edgar.setLength(myInstrument, noteDur);

```

seems to work fine. 

I also normalized wave shape so now, sine, triangle... all are the same volume and sound louder.

I will publish an update soon, right now for fun, I am trying to make an autoPlay function...

nathanRamaNoodles, 2019-05-03:
I tried your code now, and I can hear "clicks".  I liked how it sounded before.  Perhaps the full range envelopes and sine waves were necessary?  And yeah, `pause()` is broken.  Hmm, bugs ;)

wEPac, 2019-05-04:
the clicks come when a note is short and so, the rough cut will does this. We can hear them 'louder' now, just because the sound level is louder. With long notes it wont happen, the reason is, envelope is reaching the end and sound is lower. The issue is, we always had these ugly clicks and this can be solved only making an algorythm to lower down the sound level just before the note will be cut. When done, the synthetiser will make a good job ;)

The sine wave has exactly the same definition than the org code, I took care of this important point. The envelope are less precise, but we shouldnt do a difference at this point.

When normalizing sound level, I took care sound wont be satured and I adjusted each waves to give the same apparent level. If sound was satured, it would happen at the begining of the note, but it doesnt, that is well at the end of the note, this endless 'clik' issue ;)

wEPac, 2019-05-05:
yeha!!! good news, I made an autoPlay, with just few lines of code, the feeling is amazing. When running the 'partition' sketch, it will play the song without a break, while printing the music sheet! awesome

nathanRamaNoodles, 2019-05-05:
Sounds awesome man. Can't wait to hear it.

wEPac, 2019-05-05:
found a way to get ride from click. This is a common issue with most of synth, and it happens when especially with bass and with the sine wave. There are few solutions, one is to increase the wave frequency, the other one is to synch the note with the wave.

To synch will be to commplicated and to increase wave frequency like impossible. So I tried this: to generate a wave twice faster, so we re loosing precision on the sine wave itself, but the result is : hightly reduced clicks and envelope definition very much better.

It works! LOL

However, I m stepping a bit more away from the original library now :S

wEPac, 2019-05-05:
I saw something a bit boring... in casual definition of RTTL, when we have a not like "a6" it means octave 6. With your library, it means default octave + 6. I feel your way smarter, cause we can easy level up the whole song to 1 or more octave.

However the compatibility isnt preserved. I was thinking... if we defined default octave to zero, it could be interesting : 
 - if we dont define default octave in the header, so the RTTL will be compatinble (0 + 6)
 - if we define default octave, it will become (x + 6)
that way we could play casual RTTL strings, and when we want octave to increment from default, so we will define the default octave in the header.

nathanRamaNoodles, 2019-05-05:
Sounds like a good idea

wEPac, 2019-05-06:
OK, adding a double wave form is really impressive, it sounds like we are sampling at 40kHz, especially with envelopes, and particulary with the delicate sine wave. I coded it, and at the begining of the file, I added the possibility to choose  double wave or not.

However some few clicks are always here but...

I figured this, if I use create a first intance, a voice I will never use, the 2 next instances I am using to play a song wont have cliks!!! so now, it seems an issue with the synth itself. Can you help me to check for a solution?

To reproduce what I got:
 - play song `the entretainer` with sine wave and envelope 0
 - play it once with envelope 0 (it will clik)
 - play it once with envelope 1 (it wont click)
 - then modify the `table.h` file, toggle envelope 0 and 1
 - play again the song, and now, you can figure, envelope 1 will click and envelope 0 wont

adding a first voice instance will reduce this issue.

nathanRamaNoodles, 2019-05-06:
I believe the problem is the way Rttl is made.  You see, Rttl expects to run only on one instrument with no sustain.  Sustain causes one instrument to be occupied but when another note is called, then the instrument will stop playing and play the next note(which causes clicks).  The solution is to use another voice when the current instrument is occupied. This is easier to solve in MIDI because we have all instruments in one char array with multiple voices "ready" to play.
If you look at my midi example, you can get an understanding on how I use multiple voices to avoid stopping the current instrument from playing.

wEPac, 2019-05-07:
I think something is broken in your new synth file, now you are using a `numVoice++;` beside the function `edgar.setNumVoices();` you cannot anymore to create an object without to define the song, it will mess up voices order. 

Also in the interrup, here:
```

  //************************************************
  //  Modulation engine
  //************************************************
  FTW[divider] = PITCH[divider] + (int)   (((PITCH[divider]/64)*(EPCW[divider]/64)) /128)*MOD[divider];
  FTW[divider] = PITCH[divider] + (int)   (((PITCH[divider]>>6)*(EPCW[divider]>>6))/128)*MOD[divider];
tim++;
```
the two line to assign a value in ` FTW[divider]` are useless, both are doing the same thing and could be replaced by even better one  :
`FTW[divider] = PITCH[divider] + (int)(((PITCH[divider] >> 6) * (EPCW[divider] >> 6)) >> 7) * MOD[divider];`

also `tim` is not used anymore ;)

wEPac, 2019-05-07:
I see what you re explaining about Sustain and notes, but I doubt this would explain why, if I add two voices without to use them, to be neat :

this wont make clicks  :
```
MusicWithoutDelay track0a; // track we never use
MusicWithoutDelay track0b; // track we never use
MusicWithoutDelay track1;
MusicWithoutDelay track2;
```

this will make clicks  :
```
MusicWithoutDelay track1;
MusicWithoutDelay track2;
// MusicWithoutDelay track0a; // track we never use
// MusicWithoutDelay track0b; // track we never use
```

wEPac, 2019-05-07:
OK!!! I found the bug, a real big one ;) The issue is well from the synth

* org code is : 
```
  divider++;
  if (divider >= numVoice)
  {
    divider = 0;
    divider = -1;
    //tik     = 1;
    return;
  }

....

    uint8_t sample[2] = {127, 127};   // set values for channel A and B to middle of wave
    for (uint8_t i = 0; i < numVoice; i++)  // connect the voices that share the same output pin
    {
      sample[stereoMode[i]] += (_getWaveValue(i, ((byte*)&(PCW[i] += FTW[i]))[1]) * AMP[i]) >> 9; 
    }
    mOCR2A = sample[0];               // play the wave to corresponding PWM pins
    mOCR2B = sample[1];
    asm("NOP;NOP");   // Fine tuning
```
when it should be : 
```
  divider++;
  if (divider >= maxVOICES)
  {
    divider = -1;
    return;
  }

...

    uint8_t sample[2] = {127, 127};   // set values for channel A and B to middle of wave
    if (divider < maxVOICES)
    {
      for (uint8_t i = 0; i < numVoice; i++)  // connect the voices that share the same output pin
      {
        sample[stereoMode[i]] += (_getWaveValue(i, ((byte*)&(PCW[i] += FTW[i]))[1]) * AMP[i]) >> 9; 
      }
    }
    mOCR2A = sample[0];               // play the wave to corresponding PWM pins
    mOCR2B = sample[1];
    asm("NOP;NOP");                   // Fine tuning
```

Now, no more clicks and also, more important, the pitch will never increase again when adding voices!!!

wEPac, 2019-05-07:
OK I found the begining of a solution, there is something wrong in the code.

* This version will change the pitch when we re adding new voices : 
```
//*********************************************************************
//  Timer interuption
//*********************************************************************
SIGNAL(TIMER1_COMPB_vect)
{
  divider++;
  if (divider >= numVoice)
  {
    divider = -1;
    return;
  }
```

* When this one keeps same pitch what amount of voices we have, also, reducing clicks : 
```
  divider++;
  if (divider >= maxVOICES)
  {
    divider = -1;
    return;
  }
```

nathanRamaNoodles, 2019-05-07:
> if (divider >= maxVOICES)

I don't think that is a good idea because when a user wants to play 2 voices instead of the default settings of 5 voices, the user will get lag.  But I guess it doesn't matter when the user can change maxVoice settings directly.

But what happens when maxVoice is 1?  Won't there still be clipping?

nathanRamaNoodles, 2019-05-07:
The `return` is there to immediately bring the engine to the next voice.  We don't want "dead air".

wEPac, 2019-05-07:
* well if you keep `if (divider >= numVoice)` that way, the synth wont play good, like I said, it is cheating with cycle and every new voice we are adding will change the envelope of the sound. So code must be rewrote to take care of voice amount.

* the `return` is totally useful and it s providing exactly a "dead air", one sampling with nothing to happen. It should be this :
```

  divider++;
  //if (divider >= numVoice)
  if (divider >= maxVOICES)
  {
    divider = 0;
  }
```

Now I can figure, maybe the coder who made the synth added a "dead sample" to reduce click ;)

nathanRamaNoodles, 2019-05-07:
Yea, I even wonder how this piece works:
```
//************************************************
  //  Modulation engine
  //************************************************
  FTW[divider] = PITCH[divider] + (int)(((PITCH[divider] >> 6) * (EPCW[divider] >> 6)) >> 7) * MOD[divider];
```
I got it from [Dzloine](https://github.com/dzlonline/the_synth/blob/master/synth.h#L97), but I have no clue what it does.  But I went along with it and experimented.  But at least I have friends who can help me crack the code.  Alright, I'll go with your idea with getting rid of the `return statement`.

But I'm still proud of my volume control; that took months to figure out :)

wEPac, 2019-05-07:
I did few test, the `return` is a smart little thing that s reducing click. The main issue is maxVoices, really, this will be hard to make the synth to work with something else 4.

To give you an idea, the length, the pitch, the envelope, everything depend of this 5 times cycle. Also, I guess, computing a voice at each of the first 4 cycle, we must take care it will stay in the sampling interval. Computing voices one afetr the other is not purely accurate, but like we re sending notes to the synth one after the other, that works! see all things here are precise like a clock.

Dont put you down!!! You did great job with volume and sustain!!! A nice add on. And to make the synth compatible with other boards too!

You main sketch to drive the synth is also a very interesting job. There are few mibor bugs here and there and I am sure, if you had time to spend on it, you could make it perfect. You already did the hardest job!!!

When I m coding I always do 3 steps :
 - coding roughly the full project
 - looking for bugs
 - optimizing the code (faster and shorter)

Be sure the work you already did is helping me a lot, and I also learned with your code. We all have our way to make things, and that s always interesting to see how this one or this other one solved a problem.

For what I know, you re a nice person, I contacted you times ago, with a comment about one of your video, then here. You are always welcoming people and LOL, I wasnt always nice, many people wouldnt even talk about their code or what could be done.

I am sometimes a bit 'rough', especially after I spent hours to find a bug hehe, or like these days where I have strong tooth ache :P

wEPac, 2019-05-07:
I will continue and try to improove it the way I need it, maybe a bit different but close to yours, so you will be able to add the autoPlay and more to your code, even the example to write music sheet. I will also try to make 2 more envelope, we are missing ones sounding like guitar or horn. Actually I noted these settings : 
```
   Instrument         Wave Shape    Envelope
   - [organ]          SINE          0
   - [piano]          TRIANGLE      3
   - [harpsichord]    SQUARE        0
   - [trumpet]        RAMP/SAW      2
   - [clarinet]       SQUARE        1
   - [flute]          TRIANGLE      1
```

nathanRamaNoodles, 2019-05-07:
> I contacted you times ago, with a comment about one of your videos

lol, I don't remember...

> Dont put you down!!!

Don't worry.  I'm a slow-to-anger kind of guy.  I find using sarcasm is wiser 😄 .  And thank you for your kind words.  Really, I'm just happy that someone other than me is learning from my code.  It warms my heart.

> You already did the hardest job!!!

hehe, idk.  Optimizing my code down 3-4 kilobytes is more impressive.

> I have strong tooth ache

I hope you feel better.  Also, I was wondering, do you know other languages, like python and javascript?  I was making a home automation website that connects raspberry pis to esp8266 and to cell phones.  It's kinda like my other [Tinker-Controller](https://github.com/nathanRamaNoodles/Tinker-Controller).  You can join if you're interested.  I'll upload the code maybe around two weeks from now.(I have college exams next week 😀 )

wEPac, 2019-05-07:
ahah, dont miss your exams!!! the most important thing for the next of your life!

Well I like hard coding, I learn ASM years ago by myself, just for fun, but I m focusing on c++ now.

Maybe this will remain you something ;) 
https://www.youtube.com/channel/UCuTqdGMVoDy_vjyDBD8fJSA

nathanRamaNoodles, 2019-05-07:
> Maybe this will remain you something ;)
https://www.youtube.com/channel/UCuTqdGMVoDy_vjyDBD8fJSA

Link not directing me?

nathanRamaNoodles, 2019-05-07:
I found you: https://www.youtube.com/watch?v=uoHhlrqZYDI

wEPac, 2019-05-07:
strange, when I click on it, it opens youtube at the right page :S

wEPac, 2019-05-07:
lol, yes ! "Pal Treponem" a bad sexual disease

wEPac, 2019-05-08:
For what I got, a quick explanation for : 
`FTW[divider] = PITCH[divider] + (int)(((PITCH[divider] >> 6) * (EPCW[divider] >> 6)) >> 7) * MOD[divider];`

This will calculate the frequency we have to play : 
* divider: current voice we are computing
* MOD[divider]: modulation percent for the Pitch and Envelope
 - when zero => no modulation
 - EPCW[divider]:envelope ref value, PITCH[divider]:pitch ref value
* PITCH[divider]:pitch of the current voice (note)

wEPac, 2019-05-08:
published  a new version of the tool ;)

nathanRamaNoodles, 2019-05-12:
Alright, i got to testing it, and notice a few "max" variables were missing from the compiler errors.
I added them myself and got it working.  Also noticed you changed the volume channel to `CHA`(That got me for a while).
The sound quality is much better than before, and I no longer hear clicks.  Good job 👍 .
Could you explain what you did? I'm kinda interested.  Like how does this part work:
```
if (num & 0x01)       envelope[0] = (++envelope[0]) % ENVELOPES_MAX;
          if      (num == 3)    envelope[1] = envelope[0];
          else if (num & 0x02)  envelope[1] = (++envelope[1]) % ENVELOPES_MAX;
```
I noticed you did this for the sustain and waves, but how does it work? :)

wEPac, 2019-05-22:
Hello hello, still working on, but needed a rest. I simplified the code again, after several tests I noticed the main issue with click is coming when the interrupt will need more time to compute than the sample time ;) I m adding a lot of comment that will help you to understand the synth. Now it works, the last part of the job is to optimize a bit more the code and to check for timers. For the use I have, combination of SPI hardware and Music, some trouble can happen ;)

[5 more replies]