nathanRamaNoodles/Noodle-Synth #14: MIDI Support?

huantianad, 2019-08-08:
In the description it mentions MIDI files. Has this been implemented, or are you still working on it?

nathanRamaNoodles, 2019-08-08:
It supports midi commands being streamed to it.  I thought of putting the midi files directly on to it, but I thought that was beyond the scope of the project.  But if you want to play midi files, simply stream a midi file from an SD card from another arduino, and then connect the TX pin to the rx pin on the noodle synth arduino.

In the examples directory, there is a midi example that acts as a midi module.  So essentially, I would connect my arduino(teensy) to my piano or computer and stream the midi commands from a software like AnvilStudio.

huantianad, 2019-08-08:
Would it be possible to convert a midi file to the other format?

closed 2019-08-10 by huantianad