nathanRamaNoodles/Noodle-Synth #15: How to delay the music?

ahmetanbar, 2019-09-05:
Hi there, 

I try to control music by motor inputs. As music plays when motor turns, music should wait when motor don't turn. 

I try to use `pause(false)` and `pause(true)` commands, but as i see there occur a problem and the music is broken.

I know that the name of project "musicwithoutdelay" :) but i need to delay it and continue. `pause` , `resume` commands should do this, but i dont know what i miss.

nathanRamaNoodles, 2019-09-05:
Make sure you call the method `update()` in the loop.  Make sure you don't have any `delay() ` statements that block the main code from running. You can show a snippit of your code and I can see where the issue is.

ahmetanbar, 2019-09-06:
I am working on example that `Spider_Dance_Undertale_new.ino`. 

My main code is that: 

```
#include <NoodleSynth.h>
synthEngine mixer(20E3); //https://github.com/nathanRamaNoodles/MusicWithoutDelay-LIbrary
const char song[] PROGMEM = {"Spider Dance:b=115,d=16,f=abde,o=5:8f1,8c1,8a,8f,  p,c_1,8b,  b,a,e,f, p,c1,b,a,  b,c1,e,f,a,f,e,  f1,f1,e1,32c1,32b,a,  4f,4g,4a,4b,  8c1,8b,8f1,8c1,  8e#1,8d1,4c1"};
const char song2[] PROGMEM = {"Spider Dance:b=115,d=16,f=abde,o=5:8c1,8a,8f,8g,  p,a,8f,  g,f_,b,d,  p,a,f,e,  g,a,g-1,a-1,c,a-1,g-1,  a,a,8p,16p,   8p,8b-1,8p,8b-1,8p,8b-1,8p,8a-1,  8a,8d,8c1,8f,  8b,8a,4g"};
const char song3[] PROGMEM = {"Spider Dance:b=115,d=16,f=abde,o=3:f,c1,b,c1,f,c1,b,c1,d,c_1,b,c_1,c,c1,g,b,  f,c1,b,c1,  g,c1,b,c1,  a,c1,b,c1,  b,d1,c1,d1,   d,a,g,a, d,b,a,b,  d,c1,b,c1,  d,d1,c1,d1, 4c, 8p,8c,4c,  c1,d#1,e#1"};

MusicWithoutDelay buzzer(song);
MusicWithoutDelay buzzer2(song2);
MusicWithoutDelay buzzer3(song3);

void setup() {
  buzzer.begin(CHB, TRIANGLE, ENVELOPE0, 0);   //CHB means pin 3 on Arduino UNO, CHA means pin 11
  buzzer2.begin(TRIANGLE, ENVELOPE0, 0);
  buzzer3.begin(RAMP, ENVELOPE0, 0);
}

void loop() {
  buzzer.update();
  buzzer2.update();
  buzzer3.update();

  // If there is any input at A0 pin, resume the music.
  if(analogRead(A0)){
    buzzer.pause(false);
    buzzer2.pause(false);
    buzzer3.pause(false);
  }
  else{
    // Pause the music.
    buzzer.pause(true);
    buzzer2.pause(true);
    buzzer3.pause(true);
  }
}
```

As I see at original code, if buzzer, buzzer2 and buzzer3 are paused at the same time, restart the music. what is the main idea about this?

nathanRamaNoodles, 2019-09-06:
> if(analogRead(A0)){

This line may make your code too sensitive to the motor input.  Use a threshold like `if(analogRead(A0)>75)` instead.  

> As I see at original code, if buzzer, buzzer2, and buzzer3 are paused at the same time, restart the music. what is the main idea about this?

I simply wanted at least one instrument playing in a constant loop without stopping.  Then, for fun, I start all the other instruments.  (You don't have to do this)

Are you getting sound out of speaker?

ahmetanbar, 2019-09-06:
Result is broken voise. [voice from speaker](https://sndup.net/5q86) . 

I try some things like that in the loop:
```
buzzer.update();
  buzzer2.update();
  buzzer3.update();
  buzzer.pause(true);
  buzzer2.pause(true);
  buzzer3.pause(true); 
```

and  this:

```
buzzer.update();
  buzzer2.update();
  buzzer3.update();
  buzzer.pause(false);
  buzzer2.pause(false);
  buzzer3.pause(false);
```

I am surprized. There is no any voice. What is the reason of that? 

After this, I update the architecture of original code for my code: 
```
buzzer.update();
  buzzer2.update();
  buzzer3.update();

  if (analogRead(A0)>75) {
    char str = '1';
    int r;
    switch (str) {
      case '1':  //type '1' in Serial monitor.
        Serial.println("XXXXXXXXXXXX");
        buzzer.pause(false);
        buzzer2.pause(false);
        buzzer3.pause(false);
        motorMillis = millis();
        break;
    }
  }
  else if(millis()>motorMillis + 2000){
    buzzer.pause(true);
    buzzer2.pause(true);
    buzzer3.pause(true);
  }
```
The millis check just being sure that resume for 2 second at least. When I turn the motor, I see on the monitor "XXXX" text, but i can't here any voice. 

Could it be related about my circuit? I don't use any capactior or resistor?

nathanRamaNoodles, 2019-09-06:
Which Arduino are you using?  Maybe change the pin to `CHA`, pin 11.  
You might have damaged your `CHB`, pin 3, in another project(That has happened to me).

Try the [basic example](https://github.com/nathanRamaNoodles/Noodle-Synth/blob/master/examples/Basic/Basic.ino) with nothing else except for the speaker at pin `CHA`; this one works for me.

ahmetanbar, 2019-09-06:
I understand the problem. Using `pause` function endless times break the music. Music can't continue healtly way. 

Lastly my code is that: 
```
#include <NoodleSynth.h>
synthEngine mixer(20E3); //https://github.com/nathanRamaNoodles/MusicWithoutDelay-LIbrary
const char song[] PROGMEM = {"Spider Dance:b=115,d=16,f=abde,o=5:8f1,8c1,8a,8f,  p,c_1,8b,  b,a,e,f, p,c1,b,a,  b,c1,e,f,a,f,e,  f1,f1,e1,32c1,32b,a,  4f,4g,4a,4b,  8c1,8b,8f1,8c1,  8e#1,8d1,4c1"};
const char song2[] PROGMEM = {"Spider Dance:b=115,d=16,f=abde,o=5:8c1,8a,8f,8g,  p,a,8f,  g,f_,b,d,  p,a,f,e,  g,a,g-1,a-1,c,a-1,g-1,  a,a,8p,16p,   8p,8b-1,8p,8b-1,8p,8b-1,8p,8a-1,  8a,8d,8c1,8f,  8b,8a,4g"};
const char song3[] PROGMEM = {"Spider Dance:b=115,d=16,f=abde,o=3:f,c1,b,c1,f,c1,b,c1,d,c_1,b,c_1,c,c1,g,b,  f,c1,b,c1,  g,c1,b,c1,  a,c1,b,c1,  b,d1,c1,d1,   d,a,g,a, d,b,a,b,  d,c1,b,c1,  d,d1,c1,d1, 4c, 8p,8c,4c,  c1,d#1,e#1"};

MusicWithoutDelay buzzer(song);
MusicWithoutDelay buzzer2(song2);
MusicWithoutDelay buzzer3(song3);

int k = 0; 
int m = 1;
void setup() {
  // put your setup code here, to run once:
  buzzer.begin(CHA, TRIANGLE, ENVELOPE0, 0);   //CHB means pin 3 on Arduino UNO, CHA means pin 11
  buzzer2.begin(TRIANGLE, ENVELOPE0, 0);
  buzzer3.begin(RAMP, ENVELOPE0, 0);
}

void loop() {
  buzzer.update();
  buzzer2.update();
  buzzer3.update();

  // If there is any input at A0 pin, resume the music.
  if(analogRead(A0)>70){
    if(m==0){
    buzzer.pause(!buzzer.isPaused());
    buzzer2.pause(!buzzer2.isPaused());
    buzzer3.pause(!buzzer3.isPaused());
    m=1;
    }

    k = 0;
  }
  else{
    if(k==0){
      // Pause the music.
      buzzer.pause(!buzzer.isPaused());
      buzzer2.pause(!buzzer2.isPaused());
      buzzer3.pause(!buzzer3.isPaused());
      k = 1;
    }
    m = 0;
  }
}

```

Thank you for your interesting

nathanRamaNoodles, 2019-09-07:
Alright, this works for me:  (Also, I added a tempo change depending on how fast you spin the motor)

```
#include <NoodleSynth.h>    //https://github.com/nathanRamaNoodles/Noodle-Synth
#include <SensorToButton.h> //https://github.com/nathanRamaNoodles/SensorToButton
const int _threshold = 30;
const int _debounce = 100;
SensorToButton motor(_threshold, pressUp, _debounce);
synthEngine mixer(20E3);
const char song[] PROGMEM = {"Spider Dance:b=115,d=16,f=abde,o=5:8f1,8c1,8a,8f,  p,c_1,8b,  b,a,e,f, p,c1,b,a,  b,c1,e,f,a,f,e,  f1,f1,e1,32c1,32b,a,  4f,4g,4a,4b,  8c1,8b,8f1,8c1,  8e#1,8d1,4c1"};
const char song2[] PROGMEM = {"Spider Dance:b=115,d=16,f=abde,o=5:8c1,8a,8f,8g,  p,a,8f,  g,f_,b,d,  p,a,f,e,  g,a,g-1,a-1,c,a-1,g-1,  a,a,8p,16p,   8p,8b-1,8p,8b-1,8p,8b-1,8p,8a-1,  8a,8d,8c1,8f,  8b,8a,4g"};
const char song3[] PROGMEM = {"Spider Dance:b=115,d=16,f=abde,o=3:f,c1,b,c1,f,c1,b,c1,d,c_1,b,c_1,c,c1,g,b,  f,c1,b,c1,  g,c1,b,c1,  a,c1,b,c1,  b,d1,c1,d1,   d,a,g,a, d,b,a,b,  d,c1,b,c1,  d,d1,c1,d1, 4c, 8p,8c,4c,  c1,d#1,e#1"};

MusicWithoutDelay buzzer(song);
MusicWithoutDelay buzzer2(song2);
MusicWithoutDelay buzzer3(song3);
bool sync[3];
int _volume = 70;
bool _firstTime = true;
#define motorPin A0

void setup() {
  // put your setup code here, to run once:
  buzzer.begin(CHA, TRIANGLE, ENVELOPE0, 0).mute(true);   //CHB means pin 3 on Arduino UNO, CHA means pin 11
  buzzer2.begin(TRIANGLE, ENVELOPE0, 0).mute(true);
  buzzer3.begin(RAMP, ENVELOPE0, 0).mute(true);
  Serial.begin(115200);
}

void loop() {
  buzzer.update();
  buzzer2.update();
  buzzer3.update();
  setVolumeAll(_volume);
  setTempoAll(analogRead(motorPin));
  motor.read(analogRead(motorPin));
  if (_firstTime) {
    Serial.println("Pausing First Time");
    pauseAll(true);
    _firstTime = false;
  }

  // If there is any input at A0 pin, resume the music.
  if (motor.wasPressed()) {
    Serial.println("Playing");
    pauseAll(false);
  }
  else if (motor.wasReleased()) {
    Serial.println("Pausing");
    pauseAll(true);
  }
  resync();
}

void resync() {
  if (buzzer.isEnd()) {
    sync[0] = true;
  }
  if (buzzer2.isEnd()) {
    sync[1] = true;
  }
  if (buzzer3.isEnd()) {
    sync[2] = true;
  }
  int len = sizeof(sync) / sizeof(bool);
  for (int i = 0; i < len; i++) {
    if (!sync[i]) {
      break;
    }
    if (i <= len) {
      Serial.println("Re-sync");
      syncSetAll(false);
      skip(0);         //skip to beginning of song(skips in milliseconds)
    }
  }
}
void syncSetAll(bool a) {
  for (int i = 0; i < sizeof(sync) / sizeof(bool); i++) {
    sync[i] = a;
  }
}
void pauseAll(bool a) {
  buzzer.pause(a);
  buzzer2.pause(a);
  buzzer3.pause(a);
  if (!a) {
    syncSetAll(false);
  }
  else {
    long _current_time = buzzer.getCurrentTime();
    buzzer.skipTo(_current_time);
    buzzer2.skipTo(_current_time);
    buzzer3.skipTo(_current_time);
  }
}
void skip(long t) {
  buzzer.skipTo(t);
  buzzer2.skipTo(t);
  buzzer3.skipTo(t);
}
void setVolumeAll(int v) {
  buzzer.setVolume(v);
  buzzer2.setVolume(v);
  buzzer3.setVolume(v);
  buzzer.mute(false);
  buzzer2.mute(false);
  buzzer3.mute(false);
}
void setTempoAll(int t) {
  //  Serial.print("tempo: "); Serial.println(t);
  int val = map(t, _threshold, 1023, 80, 160);
  buzzer.setBPM(val);
  buzzer2.setBPM(val);
  buzzer3.setBPM(val);
}
```

ahmetanbar, 2019-09-07:
Thank you, sir. You are a whole with your project. That's perfect. Congratulations.

closed 2019-09-08 by nathanRamaNoodles