nathanRamaNoodles/Noodle-Synth #16: how do I know what note is playing now?

hdw0, 2019-12-09:
I start playing the music sequence. For every note I play, I have to light the right led. As to know what note plays in this moment?

nathanRamaNoodles, 2019-12-09:
At this [line](https://github.com/nathanRamaNoodles/Noodle-Synth/blob/master/src/MusicWithoutDelay.cpp#L378) you can get the current note playing: 
```
noodleSynth->mTrigger(myInstrument, (NOTE_C1 - 1)+((scale - 1) * 12 + note));
```
You can manually edit the library and create another private variable called `uint8_t currentNotePlaying;` and set that note as 
```
currentNotePlaying = (NOTE_C1 - 1)+((scale - 1) * 12 + note);
```
Then in your arduino code, you can do 
```
if(buzzer.currentNotePlaying == NOTE_A1){
    digitalWrite(LED_BUILTIN, HIGH);
}
```

That's just some pseudo-code though, I'll update the master-branch later.  Meanwhile, just edit the library in your own local machine.

hdw0, 2019-12-10:
Thanks very much. Thank you for your responsiveness.