nathanRamaNoodles/Noodle-Synth #18: Question: How to play MIDI files?

robatron, 2020-02-29:
From the project description:

> A User-friendly Arduino/Teensy/ESP8266 Library to play MIDI files

Sorry if I missed this, but how can one play MIDI files with this library? I saw the [How to Write Music](https://github.com/nathanRamaNoodles/Noodle-Synth#how-to-write-music) section, but didn't see anything about playing an existing MIDI file. Do the MIDI files need to be converted to this RTTL format or something?

Awesome project, BTW!

nathanRamaNoodles, 2020-02-29:
In the MIDI example listed [here](https://github.com/nathanRamaNoodles/Noodle-Synth/tree/master/examples/MIDI), you must use a MIDI serial that sends data to the RX pin of the Arduino(If you want to use it with a computer, you must use [Hairless-MIDI](https://projectgus.github.io/hairless-midiserial/)). Or you can use a Teensy as a USB-MIDI device shown at [PJRC's site](https://www.pjrc.com/teensy/td_midi.html).  Then simply plug the Arduino/Teensy to a computer and play the MIDI stream. I prefer teensy because it uses fewer wires :)

[Here is a video](https://youtu.be/dOYeKb-hXRQ?t=100) of me playing my piano and using the USB host shield to convert USB MIDI to regular MIDI and then feeding them into the RX pin of the microcontroller. You can also use [Anvil Studio](https://www.anvilstudio.com/) to play many popular midi files to the Teensy. Here is a [video of that too](https://youtu.be/uz87PWl5_uY?t=40).

I intentionally had the idea to make a library that would play a MIDI char array, but I never got around to that :sweat:.  I don't how to interpret MIDI rests and notes.  **But it is definitely possible to implement this feature**.

So that's why I've been using a separate Arduino to stream the MIDI and using my Noodle synth to play it.  

This was a hobby project back in freshmen year, so now that I look at my code, it looks so ugly XD.

robatron, 2020-03-02:
Thanks for the info! I don't know a very much about MIDI... Do you know of any other projects that can play MIDI files on an Arduino specifically? Also, could your library play MIDI files if first converted to RTTL format? Looks like there are various converters online.