nathanRamaNoodles/Noodle-Synth #19: Almost inaudible

latesh, 2020-10-06:
Thanks for making this library. Though am having one issue with it. The sound is almost inaudible on 8 ohms 0.25 watt speaker attached directly to Arduino. It's audible with [melody](https://www.arduino.cc/en/tutorial/melody) example, so the speaker is working fine! It works well when attached to desktop speakers. Is it possible to make it work on a tiny speaker? Trying to make a portable, battery operated circuit. I've tried LM386 module and that marginally increases the sound and Triangle wave seems to be the loudest. 

Any suggestions on increasing the sound output as it appears to be a software setting, considering the melody example sounds loud enough. Thanks

nathanRamaNoodles, 2020-10-06:
It's supposed to be quiet because it uses high-frequency PWM modulation.  The melody sketch you posted does not allow mixing notes together; in fact, the reason I made this library is to play multiple notes at the same time.  There's no other way to achieve this without I2S or high-frequency modulation.  You can try increasing the volume via software with [setVolume](https://github.com/nathanRamaNoodles/Noodle-Synth#new-volume-control).

I suggest getting a beefy low-cost amplifier like the [PAM8403](https://www.aliexpress.com/w/wholesale-amplifier-pam8403.html).  This is a class D amplifier (compared to the LM386 which is an AB amplifier).

latesh, 2020-10-09:
Got it! Tried PAM 8403, which induced a lot of noise. Need to figure out how to do this properly. 

Closing the issue as its not an issue with this library.

closed 2020-10-09 by latesh