nathanRamaNoodles/Noodle-Synth #21: Arduino Uno - volume problems

dwjbosman, 2020-11-23:
I can't seem to set the volume on Arduino Uno. Whatever value I set it to there is no sound output. If I leave it unchanged the sound is fine (but too loud :) )

```
// 2 channel polyphony  
MusicWithoutDelay buzzer(song);
MusicWithoutDelay buzzer2(song2);

void setup() {
  
 // buzzer2.setVolume(75); // doesn't work
 // buzzer2.setVolume(100); // doesn't work

  buzzer.begin(CHB, TRIANGLE, ENVELOPE0, 0);   //CHB means pin 3 on Arduino UNO, CHA means pin 11
  buzzer2.begin(CHB,SAW, ENVELOPE0, 0);
}

void loop() {
  buzzer.update();
  buzzer2.update();
  .....
}
```

nathanRamaNoodles, 2020-11-23:
Try putting setVolume inside the loop() after each buzzer's update().

dwjbosman, 2020-11-24:
That worked!

closed 2020-11-24 by dwjbosman