nathanRamaNoodles/Noodle-Synth #22: Odd noise during/after notes

jfbauer432, 2020-12-16:
I gave this a try but I am getting some odd noises.  Seems to be some combination of a hum and a swishing sound.  Seems to happen with/after the notes.  Here is a sample of what I am getting

:o=4,b=115:4e,2e,2p,4e,2e,2p,4e,2e,2p
Recording https://easyupload.io/76uxom

The issue is more obvious during the brief rests.  If I add additional rests I can clearly hear nothing during that time as expected.

I am using the circuit from the README.md with a LM386 based amplifier (https://www.amazon.com/gp/product/B01FDD3FYQ) and an Arduino Uno.

nathanRamaNoodles, 2020-12-16:
Try changing the frequency of the timer:
```
synthEngine mixer(20E3);
```
Instead of 20E3, use 30E3.
If that doesn't work, try changing the octave level from `o=4` to `o=5`

jfbauer432, 2020-12-16:
I was using v3.5.2, but that didn't appear to have the synthEngine stuff.  I deleted the old library and installed the latest zip (4.0.0).  Tried both synthEngine mixer(20E3) and synthEngine mixer(30E3) and I did not hear any difference.

Changing the octave made the notes sound different of course, but I still here that noise -- the pitch of the noise seems to have gone up as well.  Also tried o=6 and it was still present.